st-one-io/node-red-contrib-cip-ethernet-ip #5: NULL values on PLC reconnect

jpillman, 2018-10-31:
I'm using your module to communicate with a CompactLogix PLC and am using the "All tags" mode to return a group of tags as a single message. In a scenario where I lose communication with the PLC, the module does not function correctly when communication is restored. When communication is restored, the module begins functioning again but all values from the PLC are returned as _null._ 

The debug output from the module looks like this:
1. Communicating correctly
Debug: {"BE505Hazmon.HDSR.o_SensorData":59.67499923706055,"BE505Hazmon.HDSR.o_Fault":false}

2. Loss of communication 
Debug: Error communicating with the PLC: Error: <SCAN_GROUP> TIMEOUT occurred while writing Reading Tag Group.

3. Communication restored
Debug: {"BE505Hazmon.HDSR.o_SensorData":**null**,"BE505Hazmon.HDSR.o_Fault":**null**}

The values will continue as null until Node-Red is restarted or the flow is redeployed. 

Note: If I use the mode "All tags, one per message" this issue doesn't present itself.

gfcittolin, 2018-11-05:
Thanks for the report. I'll try to simulate it here, and eventually I may ask for more details then.

gfcittolin, 2018-11-27:
Finally I had some time to simulate the issue, and looks like it should be fixed now. Could you install the just released 1.0.1 and try it?

closed 2018-11-27 by gfcittolin

jpillman, 2018-11-27:
That fixed it! Thank you!

commit d17bd358 (2018-11-27): Fix #5: refer to controller_value when getting all values