st-one-io/node-red-contrib-cip-ethernet-ip #7: eth-ip in not working as expected

riplatt, 2019-03-19:
Hi,
I have found if you are writing a PLC variable via 'eth-ip out' and reading the same variable via a 'eth-ip in'. The 'eth-ip out' fails to update.

[image]

Node-Red reproduce: 
`[{"id":"ddbb44b9.534618","type":"eth-ip in","z":"190d888b.33b4c7","endpoint":"dd5e958.3302d68","mode":"single","variable":"Output","program":"","name":"","x":130,"y":80,"wires":[["32fe7531.aa29ea"]]},{"id":"df38996d.81d568","type":"eth-ip in","z":"190d888b.33b4c7","endpoint":"dd5e958.3302d68","mode":"single","variable":"Toggle","program":"","name":"","x":130,"y":140,"wires":[["31a8986b.ca8ca8"]]},{"id":"c0dc92b8.c7082","type":"eth-ip out","z":"190d888b.33b4c7","endpoint":"dd5e958.3302d68","variable":"Toggle","program":"","name":"","x":330,"y":200,"wires":[]},{"id":"32fe7531.aa29ea","type":"debug","z":"190d888b.33b4c7","name":"Debug Output","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":360,"y":80,"wires":[]},{"id":"31a8986b.ca8ca8","type":"debug","z":"190d888b.33b4c7","name":"Debug Toggle","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":360,"y":140,"wires":[]},{"id":"ec282ead.135fb","type":"inject","z":"190d888b.33b4c7","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":130,"y":200,"wires":[["c0dc92b8.c7082"]]},{"id":"3d3b7aa0.2d2456","type":"inject","z":"190d888b.33b4c7","name":"","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":130,"y":240,"wires":[["c0dc92b8.c7082"]]},{"id":"dd5e958.3302d68","type":"eth-ip endpoint","z":"","address":"192.168.1.10","slot":"10","cycletime":"500","name":"Test","vartable":{"":{"Output":{"type":"BOOL"},"Toggle":{"type":"BOOL"}}}}]`

Allen-Bradley test code:
`XIC(Toggle)OTE(Output);`

Node-Red Version: 0.20.0
node-red-contrib-cip-ethernet-ip Version: 1.0.1

Let us know if you need any more info or logs.

Thanks.

youngkzy, 2019-10-01:
I am working with the PCCC node for Micrologix and SLC controllers.  Now I'm testing the "Ethernet/IP" node as we have some CompactLogix controllers.   It seems to update the processor fine, but the read node does not work.   I am using a 1769-L24ER-QBFC1B controller.  How do I set up the node to read the tag value.  The Node-RED Code is simple as it reads the value, increments it and then writes it to the same tag.  I can force the tag to increment, but the input does not change state unless I restart the flow.

riplatt, 2019-10-01:
@youngkzy I found that the `eth-ip in` node will fail to update when written to from node-red but if you change the value in the controller (not through node-red) then the `eth-ip in` node will update.

As a work around for a systems where we need to read and write inputs, we make a shadow map of the inputs to read from Node-Red.

i.e.:
```
XIC(Input01)OTE(shadow_Input01)
``` 
Here you would write your values to `input01` with `eth-ip out` and read `shadow_Input01` with `eth-ip in`

Hope that helps until a real solution can be found.

youngkzy, 2019-10-03:
Thank you for your quick reply.
 I was connected to the PLC via Logix5000 software.  I updated the tag with another value.  The Node-RED input changed state!...  Each time I modified the value by means of the programming software it worked.  N70 was the first tag.  I set up 2 more tags.  N71 and N72.  Program change in Node-RED to read N71, increment and write to N70... The PLC program just copies N70 to N71.  Now it works