st-one-io/node-red-contrib-cip-ethernet-ip #8: Micro800 not established connec

lmrs10, 2019-05-03:
Hi!

I try to connect the eth-ip node to a Micro800 Allen-Bradley PLC but appear TIMEOUT ERROR and I can´t connect.

[{"id":"3eae5e2e.126f82","type":"eth-ip in","z":"afaa80e6.5121a","endpoint":"1cafa1a8.abce4e","mode":"single","variable":"Prueba1","program":"","name":"Micro870","x":170,"y":280,"wires":[[]]},{"id":"1cafa1a8.abce4e","type":"eth-ip endpoint","z":"afaa80e6.5121a","address":"192.168.1.102","slot":"0","cycletime":"5000","name":"Micro870","vartable":{"":{"Prueba1":{"type":"BOOL"}}}}]

Could you help me please?

Thanks in advanced

gfcittolin, 2019-08-21:
Sorry, I've somehow missed this issue. As far as I could read on the internet, AB Micro800 doesn't follow the exact same protocol as a ControlLogix PLC would, so I fear this is incompatible with a Micro800. Unfortunately I don't have any here so I can test. Have you tried using `node-red-contrib-pccc`?

Closing the issue for now, but feel free to reopen it's the case

closed 2019-08-21 by gfcittolin

perkasajob, 2019-08-23:
> Sorry, I've somehow missed this issue. As far as I could read on the internet, AB Micro800 doesn't follow the exact same protocol as a ControlLogix PLC would, so I fear this is incompatible with a [...]

to use micro800 seems minor,  I use https://github.com/dmroeder/pylogix , seems only handshaking header is different,

punisher97, 2019-09-21:
> > Closing the issue for now, but feel free to reopen it's the case
> 
> to use micro800 seems needs minor changes, I use https://github.com/dmroeder/pylogix , seems only handshaking header is different, I can help you to test your code, I have micro820, please find following python code snippets :
> if self.Micro800:
> ConnectionPath = [0x20, 0x02, 0x24, 0x01]
> else:
> ConnectionPath = [0x01, self.ProcessorSlot, 0x20, 0x02, 0x24, 0x01]
> and
> def _makeString(self, string):
> work = []
> if self.Micro800:
> temp = pack('<B', len(string)).decode('utf-8')
> else:
> temp = pack('<I', len(string)).decode('utf-8')
> for char in temp:
> work.append(ord(char))
> for char in string:
> work.append(ord(char))
> if not self.Micro800:
> for x in range(len(string), 84):
> work.append(0x00)
> return work

Hi, i working with Micro820, I build package [node-logix](https://www.npmjs.com/package/node-logix) i hope help to anybody

intensite, 2023-12-08:
> 
> Hi, i working with Micro820, I build package [node-logix](https://www.npmjs.com/package/node-logix) i hope help to anybody

Can msvargas's code be merged in the node-red package?