st-one-io/node-red-contrib-cip-ethernet-ip #9: Send array

GitHubAndDance, 2019-07-22:
Does node-red-contrib-cip-ethernet-ip support sending an array of data using the "eth-ip out node" to an array on the PLC, or do i need to write to each individual "subtag" with individual nodes? 

If it does support it how should msg.payload be formatted?

gfcittolin, 2019-07-30:
Good question, I don't remember trying it out with arrays. I'll check that out to see whether it works or not

gfcittolin, 2019-07-30:
Just checked out, and unfortunately there's no support on the underlying library to directly address an array, just their members. They're currently in progress to add support to this (and other features, like support for strings), so we may benefit from it in the future.

closed 2019-08-06 by gfcittolin

GitHubAndDance, 2019-08-07:
Thanks for answer