st-one-io/node-red-contrib-cip-ethernet-ip #10: TIMEOUT occurred while reading Controller Props

dirkzp, 2019-08-20:
Hi,
I'm trying to connect to an OMRON NJ101 PLC. I get the error message: "Error connecting to PLC: Error: TIMEOUT occurred while reading Controller Props.". Any idea whats wrong? IP Address is correct :-)

Thanks.
Dirk

gfcittolin, 2019-08-21:
Hi,
Never tried to connect to an Omron PLC. But from the error message, I'd say it could at least connect to the PLC, so that's good news.
The underlying library we use tries to read the controller properties just after connection (we can't control that for now), and it looks like the Omron PLC is not undestanding this request.

Unfortunately I don't have an Omron PLC here. Could you then please do a test? Please comment the [Line 125](https://github.com/cmseaton42/node-ethernet-ip/blob/master/src/controller/index.js#L125) on `node_modules/ethernet-ip/src/controller/index.js` and let me know if it works. If yes, then we can try talking fixing it upstream and have this properly fixed.

dirkzp, 2019-08-22:
Hi, 

thanks for your feedback. I've done your proposed solution. Now the connection is established. But now I'm getting the error: "Error communicating with the PLC: Error: <SCAN_GROUP>
 TIMEOUT occurred while writing Reading Tag Group." But maybe it could be a configuration issue?
Thanks.

dirkzp, 2019-10-08:
You suggested changes in the index.js is working on Line 125.
Now I configured the OMRON PLC with all tags. See attached file. Now I'm trying to read the data via Ethernet / IP with the error:  Error communicating with the PLC: Error: <SCAN_GROUP>
TIMEOUT occurred while writing Reading Tag Group."

Any suggestions how to find out the problem? Thanks. Dirk

[image]
[image]
[image]
[image]

antoniobonora, 2020-03-24:
Hi,
You must modify ethernet-ip\src\enip\cip\unconnected-send\index.js
search const build = (message_request, path, timeout = 2000) => { 
and correct to
const build = (message_request, path, timeout = 5056) => {

Bye!
Tony

dirkzp, 2020-04-23:
Hey Tony,

sorry for the late reply. I've tried your solution and it is working! 
Thanks for your help!

BR,
Dirk

closed 2020-04-23 by dirkzp

NeltherGP, 2020-06-18:
> Hey Tony,
> 
> sorry for the late reply. I've tried your solution and it is working!
> Thanks for your help!
> 
> BR,
> Dirk

Hi, I see on the screenshot that you have tags configured as Outputs on Sysmac, I also have a variable configured as an output but I get this message when trying to read from node red 
`"Error communicating with the PLC: {"generalStatusCode":30,"extendedStatus":[]}"` 
did you keep that configuration or you change something?

antoniobonora, 2020-06-24:
Hi,
I have tags input/output and not only output.
Bye!
--
Inviato da Libero Mail per Android venerdì,

Hehos12, 2020-09-14:
> Hi,
> You must modify ethernet-ip\src\enip\cip\unconnected-send\index.js
> search const build = (message_request, path, timeout = 2000) => {
> and correct to
> const build = (message_request, path, timeout = 5056) => {
> 
> Bye!
> Tony

Hi, 

I am trying your solution without succes on a Omron NX1P2. I am getting this error after modifying index.js 

"Error communicating with the PLC: {"generalStatusCode":30,"extendedStatus":[]}"

Ballupp, 2021-11-24:
> Hi, I see on the screenshot that you have tags configured as Outputs on Sysmac, I also have a variable configured as an output but I get this message when trying to read from node red `"Error [...]

Hello, I have the same issue as you after trying all the fixes above. Have you managed to solve them? Thank you!

VanOrteppi, 2022-02-11:
`"Error communicating with the PLC: {"generalStatusCode":30,"extendedStatus":[]}"` 

Struggling with the same problem. Using Omron NX1P PLC
I am able to read the data from the PLC, thanks for the good tips. However, sending Boolean does not work. I don't know of any other data types, have not tested them yet. 

[image]

I have tried to check everything. Tried to change the variables in the PLC (input/output/publish) but the error still takes place.
The tags works fine when reading them, but writing them seems to be impossible.