st-one-io/node-red-contrib-cip-ethernet-ip #11: node-red-contrib-cip-ethernet-ip returning a ""Error connecting to PLC: TypeError: "value" argument is out of bounds""

rchop3, 2019-08-26:
I'm trying to connect to a allen-bradley PLC and it works for a little bit when I reset my network but after that it shows the error "Error connecting to PLC: TypeError: "value" argument is out of bounds". Not sure why this is happening, could anyone please provide a insight please?

rafaturtle, 2019-08-29:
Hi gfcittolin,
I'm working together with rchop3.

we are just trying to read. But we are using a Siemens IOT2040, not sure if you've used that before, so the cpu is quite limited.

We are starting small, only reading 1 tag.

We've noticed that if we have a read rate of 500ms or 1000ms the connection dies pretty quickly and we can see a high CPU utilization. If we increase the read rate slowing things down, CPU gets to a confortable state, but after a few hours it still goes down.

Our issue is that it doesnt recover from that state once it goes offline, and the message is always the same "value" argument is out of bounds.

Can you help in terms of how to debug or what to look for?

thanks in advance
here is what the node looks like:

[{"id":"cf41c085.6b696","type":"eth-ip in","z":"1765960f.b6cdfa","endpoint":"77f63a2d.7631c4","mode":"single","variable":"F_N7[95]","program":"","name":"PLC Connection","x":100,"y":400,"wires":[["a9a3531d.19e6f","472b52.66d264b"]]},{"id":"77f63a2d.7631c4","type":"eth-ip endpoint","z":"","address":"192.168.1.100","slot":"0","cycletime":"15000","name":"","vartable":{"":{"CP_N199[0]":{"type":"DINT"},"F_N7[95]":{"type":"DINT"},"F_N7[1]":{"type":"DINT"}}}}]

gfcittolin, 2019-09-06:
Hi, sorry for the delay.

Regarding the device you're using, I've never personally used it, but I've heard from others that performance can be an issue for node.js applications in general, including Node-RED. (We actually developed a device that comes with Node-RED out-of-the-box as the main engine, called [ST-One](https://netsmarttech.com/page/st-one), maybe you want to take a look on that)

Regarding the main issue, there's a behavior of the node that may be influencing on this issue. Currently, when the connection drops, we reset the tag value to `null`, in the case we dropped the connection because of a badly-written value (as already happened before).

I'll take a look on the code and try to simulate this here with a SoftLogix. What controller are you trying to connect to?

rafaturtle, 2019-09-07:
Thanks for the reply.

you are right, the performance does affect the operation. So if the refresh rate is relatively high, it will loose connection and not reconnect. That was our issue.

momosjy, 2020-03-24:
I'm trying to connect to a allen-bradley PLC500 and it works for a little bit when I reset my network but after that it shows the error "Error connecting to PLC: TypeError: "value" argument is out of bounds". Not sure why this is happening, could anyone please provide a insight please?
iot2040  one tag,
[image]

abufitna, 2021-07-21:
HI.  Was using the library to connect to an L72 controllogix with a 1756-EN2T.  I am getting the following error intermittnetly....just wondering if anyone has seen this before?

21 Jul 03:14:39 - [error] [eth-ip endpoint:PLC_80] Error communicating with the PLC: Error: <SCAN_GROUP>

Unrecognized Type Passed Read from Controller: 0

``

21 Jul 03:14:39 - [red] Uncaught Exception:

21 Jul 03:14:39 - RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 3. Received 4

at boundsError (internal/buffer.js:49:9)

at Uint8Array.readUInt32LE (internal/buffer.js:181:5)

at Object.header.parse.buf [as parse] (/data/node_modules/ethernet-ip/src/enip/encapsulation/index.js:230:22)

at Controller._handleDataEvent (/data/node_modules/ethernet-ip/src/enip/index.js:240:41)

at Controller.emit (events.js:198:13)

at addChunk (_stream_readable.js:288:12)

at readableAddChunk (_stream_readable.js:269:11)

at Controller.Readable.push (_stream_readable.js:224:10)

at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)

npm ERR! code ELIFECYCLE

npm ERR! errno 1

``

npm ERR! node-red-docker@1.3.5 start: node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS "--userDir" "/data"``

npm ERR! Exit status 1