st-one-io/node-red-contrib-cip-ethernet-ip #16: TIMEOUT occurred while reading Controller Props

SrSeraph, 2020-02-16:
Hi,

I'm trying to use connect to a Micro850 (2080-LC50-48QBB) but i always get the controller props timeout.

I think I'm connecting properly, the error message is different if I use a other IP's than the Micro850, which means I'm a step closer to get it working.

I have Local and Global variables configured in the controller which is running.

Does the Address also needs Port or a reference to the Backplane?
Or do i need to Enable CIP in the controller parameters?

Anyone knows why i'm having this problem?

Thanks

gfcittolin, 2020-02-17:
As far as I'm aware of, the library we (ethernet-ip) use does not support the Micro800 series (yet). There are some minor protocol differences that make the communication setup incompatible. We need to wait until it's supported there so we can also support it here too.

bkvargyas, 2020-03-15:
Hello -- I'm trying this module with Automation Direct Click PLC.   They added support for Ethernet/IP around August of 2019.   I'm also getting the timeout issue.    Would be great to support this low cost PLC, It's under $200.

danvandam01, 2022-09-23:
any update on the compatibility with Micro800 family of PLCs?