st-one-io/node-red-contrib-cip-ethernet-ip #19: cip

sujayuid, 2020-03-22:
"Error connecting to PLC: Error: TIMEOUT occurred while reading Controller Props."

any one have solution.

gfcittolin, 2020-05-24:
Hello, which controller (model and part-number) are you trying to connect to?

xor0110, 2020-07-17:
I am having a similar issue.  I am trying to read a single global tag from a Micro820.  Single tag read works fine with the ControlLogix controller.  I believe the Micro800's are a bit different.  Can you confirm this node isn't intended to be used with the Micro800's?  Thanks.

gfcittolin, 2020-07-20:
Exactly. Unfortunately the connection process on Micro800 are a bit different than what we normally find on ControlLogix/SoftLogix. As far as I read about on the internet, they are not that much different, but the library we use does not support that yet, and we don't have one so that we could develop/test this.

CryoBuilt, 2021-02-19:
> Exactly. Unfortunately the connection process on Micro800 are a bit different than what we normally find on ControlLogix/SoftLogix. As far as I read about on the internet, they are not that much [...]

If I were to provide a micro800 PLC would you be able to get this supported?

dpatel2016, 2021-12-13:
Is there any possibility of supporting Micro800s anytime soon? We have been using both Micro800s and MicroLogix 1100, but MicroLogix 1100s are getting discontinued next year so we are migrating to all Micro800s for any new projects.