st-one-io/node-red-contrib-cip-ethernet-ip #28: Error communicating with PLC, Omron NX1P

VanOrteppi, 2022-02-15:
I'm having a problem when trying to write to Omron PLC

It's somewhat the same issue descripted [here,](https://github.com/st-one-io/node-red-contrib-cip-ethernet-ip/issues/10]) but this occurs when writing data to PLC in Node Red. After commented the line 125 as instructed to do in [https://github.com/st-one-io/node-red-contrib-cip-ethernet-ip/issues/10](url) I am able to read the data from the PLC without problems.

I just noticed that this thread was already closed so probably nobody will see my message there...

So anyway, when I try to send any data from Node Red to the PLC with the "eth - ip out" -function, the status of the function block changes (for example from false to true), but after a few seconds the function block goes in error and the error message in debug window tells the following: 

[image]

"Error communicating with the PLC: {"generalStatusCode":30,"extendedStatus":[]}"

Is there anything I could check or do differently?

caoxi1982, 2022-06-08:
any update it ?,I have the same issue , with L340ERM,and at the beginning some minutes I saw the correct data , but later it always error communicating with the PLC ,even I start/stop PLC

MountainKing91, 2022-07-24:
To the best of my knowledge, these EIP nodes do not work with Omron NX/NJ PLCs right out of the box; you need adjust the timeout in `\ethernet-ip\src\enip\cip\unconnected-send\index.js`:
[image]

VanOrteppi, 2022-07-25:
Thank You for the tip. Unfortunately this didn't work out. This setting was already set to 5056, and I think that it should have worked with the previous setting.
Off course I tried this new timeout setting as well, but the result is still the same. 
Still looking the solution for this.