st-one-io/node-red-contrib-cip-ethernet-ip #39: Plc is not automatically reconnected

bombjackit, 2023-06-29:
### Describe the bug
When a plc, for a known reason (like maintenance, firmware upgrade or software upload), is disconnected from the lan, sometimes it does not automatically reconnect : I have to restart flow manually.

### Expected behavior
I expect that the plc is reconnected, in all case, automatically

### Logs
This is the log (after this plc is disconnected)
26 Jun 08:03:27 - [error] [eth-ip endpoint:Plc] Error communicating with the PLC: {"generalStatusCode":4,"extendedStatus":[0]}

### Environment
 - node-red-contrib-cip-st-ethernet-ip 2.0.0-beta.3
 - Node red 2.2.2
 - Node.JS 14.21.1
 - Plc Rockwell 1756L81ES
 - Windows 10 professional 21H2

r4nd0wn, 2023-07-03:
same on a linux machine. This error occurs every time there is a download on the plc side and is elimited as soon as I redeploy (probably because of the forced reconnect).

cgraun, 2023-08-03:
Same here.

### Describe the bug
After a network interruption (unplugging PLC network cable) or going to STOP and RUN it automatically reconnects. However, after downloading the program, it doesn't, throwing the following error on console.
The only way to get it back working I've found is restarting node-red.

### Logs
3 Aug 10:48:12 - [error] [eth-ip endpoint:myPLC] Error communicating with the PLC: {"generalStatusCode":4,"extendedStatus":[0]}

### Environment

- node-red-contrib-cip-st-ethernet-ip 2.0.0-beta.3
- Node red 2.2.2
- Node.JS 16.15.1
- Plc Rockwell 1756L62S
- Windows 10 Enterprise 2016 LTSB

bombjackit, 2024-01-30:
Reposted in the right place https://github.com/SerafinTech/node-red-contrib-cip-st-ethernet-ip/issues/5

bombjackit, 2024-02-21:
@cgraun and @r4nd0wn there is a new version of node-red-contrib-cip-st-ethernet-ip (2.0.2, see https://www.npmjs.com/package/node-red-contrib-cip-st-ethernet-ip) that may resolve this issue (I try thid new version and works well but actually I haven't try if auto reconnect in case of download really works).

bombjackit, 2024-03-04:
I tried reconnect after download to plc (with node-red-contrib-cip-st-ethernet-ip 2.0.2) and all go well.