st-one-io/node-red-contrib-cip-ethernet-ip #40: Some Plcs going offline after updating st-ethernet-ip

bombjackit, 2023-07-03:
### Describe the bug
After updating library st-ethernet-ip (used by _node-red-contrib-cip-st-ethernet-ip 2.0.0-beta.3_) from 2.5.3 to 2.6.1 some plcs was signed as offline by _eth ip in_, but this is not true (all work well).

### How to solve problem
Going deeper thru the problem, I focused on a PLC from which I read only one element x.y[50] : x is a struct, y is an array of 64 DINT, it was enough to change type from DINT to empty (in plc tags configuration on nodered), after deploy plc was signed online and all go well.

### Expected behavior
_Eth ip in_ must not sign offline a plc that work and that is well configured.

### Environment
 - **Version** of this node: node-red-contrib-cip-st-ethernet-ip 2.0.0-beta.3
 - **Node-RED version**: 2.2.2
 - **Node.JS** 14.21.1
 - **PLC Part Number**: 1756-L83ES
 - **OS**: Windows 10 professional 21H2

zenovix, 2023-07-05:
i was facing the same problem, then i read all struct using data type Struct, the output type was a object then i filter it out of my requirement.

neben, 2023-09-18:
I'm facing a similar issue, where the struct element is read only once after restarting the flow, and then nothing. Reading the entire struct works though.

bombjackit, 2024-01-30:
Reposted in the right place https://github.com/SerafinTech/node-red-contrib-cip-st-ethernet-ip/issues/4