st-one-io/node-red-contrib-cip-ethernet-ip #43: "Error communicating with the PLC: {"generalStatusCode":1,"extendedStatus":[273]}"

Yusseiin, 2023-08-30:
### Describe the bug
I am not able to connect to any PLC.
I am connected to a 1756-ENBT/A, 192.168.11.204 slot 01, and i want to comunicate to a PLC L81ES, slot 02.
[image]
But it is not connecting saying `"Error communicating with the PLC: {"generalStatusCode":1,"extendedStatus":[273]}"`
I have tryied with another PLC but i got the same error.

### Expected behavior
Connect, i have tryed with PCCC and it is working fine i am not able to understand why it is not working

### Screenshots
[image]

[image]

### Environment
 - **Version** of this node: [2.0.0-beta.3]
 - **Node-RED version**: [3.0.2]
 - **PLC Part Number**: [L81ES ]
 - **OS**: [Ubuntu]
 - **Browser**: [Firefox]

cosminbulai, 2023-08-31:
I'm also experiencing this issue with the same version of the node [2.0.0-beta.3].

Node-RED version v3.02. CompactLogix 5370, Ubuntu, Chrome

jamesdebono, 2023-12-13:
I get the same error on 2.0.0-beta.3.
1756-L83ES slot 0, 1756-EN2T slot 1.
Old version 1.1.3 works.

AlexOrmanBayer, 2024-01-04:
Getting the same error. 
Seems like the enum/offset/something for the types are wrong in version node-red-contrib-cip-st-ethernet-ip@2.0.0-beta.3

While lacking a bunch of types, I'm seeing success with node-red-contrib-cip-ethernet-ip@1.1.3

bombjackit, 2024-01-30:
For issue regarding node-red-contrib-cip-st-ethernet-ip (2.0.0.0) you the right place to post is https://github.com/SerafinTech/node-red-contrib-cip-st-ethernet-ip/issues

Yusseiin, 2024-01-30:
> For issue regarding node-red-contrib-cip-st-ethernet-ip (2.0.0.0) you the right place to post is https://github.com/SerafinTech/node-red-contrib-cip-st-ethernet-ip/issues

Posted even on the other github, hope for an answer

[https://github.com/SerafinTech/node-red-contrib-cip-st-ethernet-ip/issues/6](url)

bombjackit, 2024-01-30:
I request to SerafinTech to add issue button to his fork, he kindly added this feature today.

I hope that all issue will have a positive response.

https://github.com/SerafinTech/node-red-contrib-cip-st-ethernet-ip/issues/6<https://urlsand.esvalabs.com/?u=http%3A%2F%2Furl&e=7a7bd34c&h=f36dc151&f=y&p=n>

Messaggio analizzato da Libraesva ESG.
Segnala come malevolo/spam.<https://alfaacciai.esvacloud.com/action/4TPHMT3P9xz7t8q/submit-as-bad>
Mettilo in blocklist.<https://alfaacciai.esvacloud.com/action/4TPHMT3P9xz7t8q/blocklist>